inib/Marlin #8: BootScreen stuck on TFT Updating 100% after flashing

slmmnl, 2019-11-25:
### Description
Screen is stuck and cant proceed further. 

### Steps to Reproduce
Cloned TFT+Probe 
Compiled in PlatformIO

**Expected behavior:** Usable Display, usable printer
**Actual behavior:** Frozen on TFT Updating 100% (Restarting makes it say Booting...; but never boots)

#### Additional Information

Config.h as is in the repo. No changes

dreamer2, 2019-11-26:
check your sources, similar problem was on 4.1 platformio update and was fixed here https://github.com/inib/Marlin/commit/0672e7e82c0f34436978a1d928b9dbd87ce93ef6

dreger1123, 2019-11-27:
Having same problem please help. running the newest version of platformio

dreger1123, 2019-11-27:
> Having same problem please help. running the newest version of platformio
> 
> Error Message however it looks like a warning and not an error, it still spits out robin_nano.bin
> 
> *** [.pio\build\mks_robin_nano\firmware.bin] TypeError : a bytes-like object is required, not 'str'
> Traceback (most recent call last):
> File "C:\Users\Dreger Dietle.platformio\packages\tool-scons\script..\engine\SCons\Action.py", line 1209, in execute
> result = self.execfunction(target=target, source=rsources, env=env)
> File "C:\Users\Dreger Dietle\Desktop\Marlin-2.0.X-SapphirePro-3.5TFT-Probe\buildroot\share\PlatformIO\scripts\mks_robin_nano.py", line 25, in encrypt
> robin.write(byte)
> TypeError: a bytes-like object is required, not 'str'

UPDATE:
 I got it to compile successfully by replacing the script file with the one mentioned above however it still gets stuck at boot, It says TFT Updating 100% then reboot and it just gets stuck at booting... screen

dreamer2, 2019-11-27:
try build from SapphirePro-3.5TFT tree

inib, 2019-11-27:
https://github.com/inib/Marlin/tree/2.0.X-SapphirePro-3.5TFT-Probe is not up to date and does not have the latest fixes for python3 and platformIO.

Please use the main branch https://github.com/inib/Marlin/commits/2.0.X-SapphirePro-3.5TFT and apply the changes for a probe there ( https://github.com/inib/Marlin/commit/3b6ac0d8a8015642134cd7f961db743e21bebc3f )

dreger1123, 2019-11-27:
> https://github.com/inib/Marlin/tree/2.0.X-SapphirePro-3.5TFT-Probe is not up to date and does not have the latest fixes for python3 and platformIO.
> 
> Please use the main branch https://github.com/inib/Marlin/commits/2.0.X-SapphirePro-3.5TFT and apply the changes for a probe there ( [3b6ac0d](https://github.com/inib/Marlin/commit/3b6ac0d8a8015642134cd7f961db743e21bebc3f) )

I already have started working on changing the main branch to work with the BLTouch I cannot figure out what the servo PIN number is for the built in BLTouch plug on the board, let me know if you guys do. I think it’s 16?

slmmnl, 2019-12-02:
The main branch works fine, got the probe working through the 3.5TFT repo. 

Thanks again

closed 2019-12-02 by slmmnl