inib/Marlin #9: Bilinear bed leveling fails at last row

migger80, 2019-11-28:
First, thank you for making these tweaks. It has made my Sapphire Pro soo much nicer.

### Description

When enabling BLTouch probe and AUTO_BED_LEVELING_BILINEAR the last row of points is skipped.
I guess it has been fixed in the main Marlin, is it possible for you to incorporate the fix in your version?
https://github.com/MarlinFirmware/Marlin/pull/16018

I don't know exactly how github works, and if this is something I can do myself. But I don't know where to start.

inib, 2019-11-30:
Yes, I will try to update this fork as soon as I have time to test it.
until then
- You can take a look at the changes made and apply them manually: 
https://github.com/MarlinFirmware/Marlin/pull/16018/files
- You can pull the latest marlin bugfix branch on your local repo and get all the latest commits: 
https://help.github.com/en/github/using-git/getting-changes-from-a-remote-repository

rt1970, 2020-03-03:
Hi there!
I have the same problem! G29 stopped at 6/9
Is there already a solution?