inib/Marlin #10: Extruder motor rattling when Linear Advance is enabled.

porlock, 2019-11-30:
I tried many combination of jerk and acceleration but no luck :( I testes already 3x Nema 17 motors and 2x a4988 step drivers, no luck

https://youtu.be/OAC048VGxtk

[config.zip](https://github.com/inib/Marlin/files/3906586/config.zip)

inib, 2019-11-30:
Does rattle mean you are losing steps and is it doing that on a retraction move?

With linear advance enabled, I'm fine with only 1.6-2mm retraction.

porlock, 2019-11-30:
it not seems to loosing steps, and it not happens on retraction but when acceleration occurs, 
Check my movie, extruder motor is unmounted from extruder so there is no force on it and still ratling.

inib, 2019-11-30:
This might also depend on the slicer you are using, slic3r/PrusaSlicer resets the jerk/acceleration settings in every gcode file no matter what you have configured in the firmware.

best way to tune those settings is via gcode while printing over serial. If that doesn't help, try deactivating "s-curve acceleration"

MH-GoL, 2019-11-30:
Sounds like "normal" stepper sounds on slow rotations for me. Try to use microstepping or change to a tmc driver for smooth drive

porlock, 2019-12-01:
It is not normal it is not my 1st printer im 100% sure, what i notice is it rattling only when junction deviation is disabled, on enabled everything is find and it is depended of mks robin nano board version.
mks robin nano 1.1 = rattling when LA i JD enabled
mks robin nano 1.2 = rattling when LA enabled and JD disabled

What version of MOBO do you have, and what is yours LA i JD settings ?

inib, 2019-12-01:
I have the 1.2 Board and I usually use
LA K=0.7
JD K=0.02 for first layer and K=0.15/K=0.3 for normal and fast settings for the actual print. (which is sometimes limited by my extruder movements, which leads to some rattles)
I also use a TMC2130 on the extruder motor in 1/16th spreadcycle with interpolation.

porlock, 2019-12-01:
Did you tried LA enabled and JD disabled ?
Im wondering how do you figure out to use different K on other layers ?

sobieh, 2019-12-02:
it is caused probably by a Marlin bug when you enable CLASSIC_JERK + LIN_ADVANCE combination.
It produces an undefined behavior with undefined MAX_E_JERK value.

I'll check that to be sure and make a PR fixing it later.

inib, 2019-12-02:
In PrusaSlicer I have
`M900 K0.7 ; K-factor
M205 J0.02 ; reduced K-factor for 1st Layer`
in my filament g-code settings and
`{if layer_num == 1}M205 J0.3{endif} ;`
in the _after layer change_ gcode in the machine settings.

PrusaSlicer also resets the machine limits in each gcode file, this could lead to the different behaviors, if there is somewhere a bug related to that.

sobieh, 2019-12-02:
Ok they already fixed it on the official marlin repo.
The only thing you need is to pull their changes :)