inib/Marlin #15: Printer freezing after some time

Technology89, 2020-01-18:
I've uploaded your firmware and everything seemed to be fine. But after ~1h the printer just freezes. LCD not working anymore and whole printer stands still. After a reboot everything seems normal. When i then start another print it freezes again after some minutes. When i let it sit for a while it crashes after ~1h again.

GeraltR, 2020-01-18:
I have the same problem, but my Sapphire freezes every now and then. Additionalas every now and then the hotend goes to the maximum position in the X or Y axis and goes back to printing. The latter case is better because the printout is not completely broken.

novio8, 2020-01-18:
I have the same problem, similar simptoms

Technology89, 2020-01-18:
Good to see that I am not the only one. I'm gonna roll back to 320 pixel version with original branch. Hopefully the 480 gonna be implemented soon into offical branch.

novio8, 2020-01-18:
Technology89, which offical Branch you are using ?
And you had no issues with it ?

atomphil, 2020-02-03:
I got issues like GeraltR too. I researched a little bit and now I´m sure, the cause are errors while reading from sdcard. Here I opened a thread in German: https://drucktipps3d.de/forum/topic/sapphire-pro-mks-robin-nano-druckkopf-faehrt-irgendwohin/

Here is a related issue on github: https://github.com/MarlinFirmware/Marlin/issues/16672

GeraltR, 2020-02-05:
Hi! I've read the threads with similar problem indicated by atomphil and unfortunately none of the solutions given fixes the problem. I'm hooking up to the Repetrier-Host printer to keep track of what's going on. When trying any printout from the card, the errors monitored by Marlin M27 come out.  The errors are two sometimes three per second. 
I changed and formatted the cards. 
My problem came after installing the tangent version 2.0.
According to me this is not a card problem because the same file from the card printed with the original FW does not send out errors.

Translated with www.DeepL.com/Translator (free version)

GeraltR, 2020-02-10:
[https://github.com/MarlinFirmware/Marlin/issues/16672#issuecomment-583304685](url)